TerosTechnology/vscode-terosHDL #353: csv import not working

plsecker, 2022-11-24:
I cannot import a csv file, even a simple file.  Tried multiple options eg. with quotes around the file name. 
Manual file import works ok.
No information as to why not.  Be good if there was a log file.

VSCODE: Version: 1.73.1
Commit: 6261075646f055b99068d3688932416f2346dd3b
Date: 2022-11-09T03:54:53.913Z
Electron: 19.0.17
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Linux x64 5.15.0-53-generic
Sandboxed: No

qarlosalberto, 2022-11-24:
Could you share your csv file?

Yes, I need to create a log file.

plsecker, 2022-11-24:
[test.csv](https://github.com/TerosTechnology/vscode-terosHDL/files/10086996/test.csv)

```
philip@philip:/$ ls /media/sf_Vbox/arty7_PI_4377/viv3/srp3.gen/sources_1/bd/rxFoc_ip/synth/rxFoc_ip.vhd
/media/sf_Vbox/arty7_PI_4377/viv3/srp3.gen/sources_1/bd/rxFoc_ip/synth/rxFoc_ip.vhd

```

If I put the files in quotes with a comma something gets imported (library?) see odg file

[screen.odg](https://github.com/TerosTechnology/vscode-terosHDL/files/10087020/screen.odg)

qarlosalberto, 2022-11-25:
Could you send me an email to: carlosruiznaranjo@gmail.com?

plsecker, 2022-11-26:
Hi!

Keen to user this tool.

Philip

qarlosalberto, 2022-11-26:
With your email, please.