TerosTechnology/vscode-terosHDL #355: VHDL LS does not work with a project where only the run.py of VUnit is added

Vallet13, 2022-11-28:
When creating a project with a VUnit run.py as only file VHDL LS does not seem to recognize other files of the project when linting. 

The generated teroshdl_export.json indicates the file is included:

>     "files": [
>         {
>             "file_name": "/home/user/Documents/testing/custom/test_pkg.vhd",
>             "library_name": "lib"
>         },

However, VHDL LS reports:

> No primary unit 'test_pkg' within library 'work'

This goes away when adding the file to the project manually.

 - OS: Ubuntu 20.04
 - VSCode version 1.73.1

Vallet13, 2022-11-28:
That's exactly what I would like to happen and expected to happen when reading "If you are using VUnit you only need to add the run.py file." in the 'Adding files' section pf 'Project manager' in the documentation.

It would make adding files to the project quicker and easier when using VUnit anyway.

Vallet13, 2022-11-28:
Wasn't meaning to close this with the comment, sorry.

closed 2022-11-28 by Vallet13

qarlosalberto, 2022-11-28:
I will add it as future feature :)