TerosTechnology/vscode-terosHDL #356: File paths need to be escaped for special characters

holzi-x, 2022-11-29:
**Describe the bug**
On Windows, Teros does misinterpret project file paths containint "\t". File references with "\t" need to be escaped with "\\t". IMHO this should be done by teros when saving the file to the project yaml.

**To Reproduce**
` - name: "test\\tb_CarrierGen.vhd"
    file_type: "vhdlSource-2008"
    is_include_file: false
    include_path: ""
    logical_name: ""`

**Please complete the following information:**
 - OS: Windows 10 Pro 21H2
 - VSCodium V1.73.1

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.

gmartina, 2024-04-17:
PR #594 changed the way paths are saved in windows using forward slash instead of back slash. This bug should be fixed. cc: @qarlosalberto 

Example:
```
  - name: "src/tfsm_ex.vhd"
    file_type: vhdlSource
    file_version: 2008
    is_include_file: false
    include_path: ""
    logical_name: lib1
    is_manual: true
    source_type: none
```

qarlosalberto, 2024-04-17:
thanks, closing issue

closed 2024-04-17 by qarlosalberto