TerosTechnology/vscode-terosHDL #357: Linting VHDL not finding libraries in test bench file with Vunit

altec921, 2022-12-07:
When writing a test bench with Vunit in the IDE, linting of the the Test bench file does not seem to work. The linter is unable to find the vunit_lib or the modules that I am trying to test. 

VHDL LS reports: 
> No such library 'vunit_lib'vhdl ls

The TerosHDL project contains a run.py and two vhd files, the module and the test bench for it.
The run.py is set as the top file. Running the project works successfully and the tests pass. 

Linting works on the main file. Only the testbench seems to be the issue. 

the teroshdl_export.json has all the source files and vunit_lib files listed and the names match up. to what is in the testbench

> ```{
>             "file_name": "/home/user/.local/lib/python3.8/site-packages/vunit/vhdl/verification_components/src/axi_pkg.vhd",
>             "library_name": "vunit_lib"
>         },
>         {
>             "file_name": "/home/user/.local/lib/python3.8/site-packages/vunit/vhdl/string_ops/src/string_ops.vhd",
>             "library_name": "vunit_lib"
>         },
>        ... etc
> ```

OS: Ubuntu: 20.04
VSCode version: 1.73.1
TerosHDL Version: 2.08

qarlosalberto, 2022-12-07:
Related to #355

altec921, 2022-12-08:
@qarlosalberto I had a look at #355. 

I think my issue is a little different as The test bench file is added to the terosHDL project manually. So from what i understand the file should still be linted as it is included in the project.