TerosTechnology/vscode-terosHDL #358: [Verilator] IMPLICIT warnings not appearing

michael-etzkorn, 2022-12-08:
Perhaps this is a bug on my end, but if I have implicits connected to a module 

```verilog

module dummy;
  dummy1 d1 (
    .implicit(implicit)
  );
  dummy1 d2 (
   .implicit(implicit)
  ); 
endmodule 
```

verilator will throw an IMPLICIT warning on linting, but this does not appear in VScode.

qarlosalberto, 2022-12-08:
Cool, I would need a minimum reproducible example.

Also, could you share the Verilator output? So I can check if there is a parser bug

michael-etzkorn, 2022-12-08:
When I manually run verilator, I see several statements like this: 

```
foo/bar.v:493:31: Signal definition not found, and implicit disabled with `default_nettype: 'baz'
  493 |       .baz         (baz),
```

I don't believe it's a parsing issue, but I'm not sure how to see what TerosHDL is outputting. It may have something to do with maximum number of errors? Investigating now.

qarlosalberto, 2022-12-08:
I use:

```
verilator --lint-only -sv -Wall -bbox-sys --bbox-unsup -DGLBL file.v
```

qarlosalberto, 2022-12-08:
In Vscode go to: help, developer tools, console. You will se the command

michael-etzkorn, 2022-12-08:
Ok apparently, it's an issue with lower-level modules causing verilator to terminate early which I wasn't seeing. 

If I add `-Wno-TIMESCALEMOD` and `--error-limit 10000`, I can now see the implicit warnings show up. 

Thanks for pointing me to the developer settings! None of what Teros passes was problematic, I just wasn't including my full include list when running in a local terminal so I didn't see the TIMESCALEMOD errors were terminating verilator early.

closed 2022-12-08 by michael-etzkorn