TerosTechnology/vscode-terosHDL #360: Else statement indentation with VHDL formatter

dubicube, 2023-01-02:
When using the VHDL standalone formatter, "else" and "elsif" statements are indented one step too far to the right:
[image]
As I understand how the indentation currently works, it takes all the code between "if" and "end if;", and add a tab to this block.
Instead, I would expect that "else" and "elsif" statements should be at the same indentation level than the "if" and "end if;", like bellow:
[image]

Am I the only one who expects this behaviour ?

**Versions:**
 - Windows 10
 - VSCode version 1.74.1

**Formatter settings**
[image]
Indentation is configured with 3 spaces (please consider I am not responsible for this choice...).

qarlosalberto, 2023-01-02:
It's a bug.

qarlosalberto, 2023-01-02:
TerosHDL shouldn't allow 3 spaces 😜

dubicube, 2023-01-03:
I reproduced the bug.
It happens with a special file. When I invoke the formatter on this file, the formatter does not work (it does nothing). But after this try, the formatter will produce the indentation bug previously described in every other files. I need to restart vscode to get back to a normal behaviour.

I truncated the file generating the bug to a minimalist size. You should be able to reproduce the bug with it. Here it is:

library IEEE;
use IEEE.STD_LOGIC_1164.ALL;

use IEEE.NUMERIC_STD.ALL;

entity top is
end top;

architecture STRUCTURE of top is

   type t_v32_array is array (natural range<>) of std_logic_vector(31 downto 0);
   constant c_CONSTANT_REGISTERS : t_v32_array(0 to 1) := (
      x"00010203",
      x"00010203"
   );

   component yousk2 is
   port(
      CLK : out std_logic
   );
   end component yousk2;

   signal CLK : std_logic := '0';
   signal s_counter : unsigned(7 downto 0) := (others => '0');

begin

   process(CLK) begin
      if rising_edge(CLK) then
         if (s_counter >= 42) then
            s_counter <= s_counter + 1;
   else
      s_counter <= (others => '0');
         end if;
      end if;
   end process;

end STRUCTURE;