TerosTechnology/vscode-terosHDL #367: Brackets in VHDL IF statement will prevent the state machine viewer from proper visualization

Dawncrasher, 2023-01-12:
Hi fellows,

using brackets in the if statement of VHDL code will hinder the state machine viewer to properly resolve the connections.
Following code snippet will visualize the following picture.
`            case state is
                when S_INIT =>
                    state <= S_IDLE;
                when S_IDLE =>
                    if condition then
                        state <= S_PROCESS_MESSAGE;
                    end if;
                when S_PROCESS_MESSAGE =>
                    if (condition) then
                        state <= S_IDLE;
                    end if;
                end case;`
[image]

Following code snippet will visualize the following picture.
`            case state is
                when S_INIT =>
                    state <= S_IDLE;
                when S_IDLE =>
                    if condition then
                        state <= S_PROCESS_MESSAGE;
                    end if;
                when S_PROCESS_MESSAGE =>
                    if condition then
                        state <= S_IDLE;
                    end if;
                end case;`

Hopefully this helps. Thanks for creating terosHDL!

Best Regards,

Johannes

qarlosalberto, 2023-01-12:
Hi @Dawncrasher , thanks for the report.

Could you send me an email to carlosruiznaranjo@gmail.com I would like to share a beta version.

qarlosalberto, 2023-06-26:
fixed on v5.0.0

closed 2023-06-26 by qarlosalberto