TerosTechnology/vscode-terosHDL #373: hdlmake projects

Qbort, 2023-02-09:
I really like teroshdl and wanted to ask if you had considered supporting hdlmake projects (see https://hdlmake.readthedocs.io/en/master/)

It would be really nice to not have to also create a TerosHDL project when I already have one in a vendor agnostic format.

SenorMagooo, 2023-02-17:
Great idea. Can you please label this issue as "enhancement" so we can track it.

Qbort, 2023-02-23:
Sorry I don't think I have the permissions to add labels to this issue.

smgl9, 2023-02-24:
Some TerosHDL features rely on edalize. As far as I know hdlmake have many overlaps with edalize. 
Anyway we can investigate how to use hdlmake files inside TerosHDL. Can you provide some examples of hdlmake?

Qbort, 2023-03-01:
The majority of CERN open hardware projects are made with hdlmake: https://ohwr.org/project

The simplest example is probably the one provided with hdlmake itself: https://hdlmake.readthedocs.io/en/master/#learn-by-example

More complex projects, see: https://hdlmake.readthedocs.io/en/master/#advanced-examples

Thanks for the information on edalize - I did not know about it.