TerosTechnology/vscode-terosHDL #379: contribute zeal/dash document

wieniawski-lai, 2023-02-24:
hi,

I have generate a zeal/dash document for TerosHDL.

[image]

which is an offline help document of HDL.

Is there a way to upload it to public?

qarlosalberto, 2023-02-24:
This is the repo of the new documentation: https://github.com/TerosTechnology/teroshdl-docusaurus-doc

Could you do it in that repo?

wieniawski-lai, 2023-02-24:
I create the docset in a tedious way:
1. download the [helpfiles](https://terostechnology.github.io/terosHDLdoc/) to offline.
2. use [html2dash](https://github.com/wieniawski-lai/html2Dash) to generate the docset.
3. where [html2dash](https://github.com/wieniawski-lai/html2Dash) is folked from [selfboot](https://github.com/selfboot/html2Dash) and updated.

I donot know this [document repo](https://github.com/TerosTechnology/teroshdl-docusaurus-doc) before, so my work is not simplified.

It is better to generate docset from the original md file in that repo instead of downloaded offline html file. Which I will find how to later.

And the docset is large, about 30M byte.
[image]

need I check in the docset?

or just check in the html2dash python file?

closed 2023-06-22 by qarlosalberto