TerosTechnology/vscode-terosHDL #381: [BUG] Schematic Viewer Failed when using Yosys + GHDL(module) configuration

kevingoh, 2023-03-10:
**Describe the bug**
When trying to generate schematic of my designs in VHDL, which are synthesised successfully using GHDL, it failed when trying to generate schematic with the Yosys command generated. The error was asserted by GHDL plugin module because the TOP module isn't specified in the Yosys command generated by the tool.

Sample of the command generated by terosHDL:

yosys -m ghdl -p "ghdl --std=08 -fsynopsys top.vhd -e; proc; opt; write_json teroshdl_yosys_output.json; stat",

The TOP entity in top.vhd is supposedly specified after the -e switch in the command. The top.vhd was selected as as the top in the tool.

**To Reproduce**
By selecting "show the netlist" with Yosys + GHDL(module) configuration on top module written with VHDL.

**Code**
N/A

**Please complete the following information:**
 - OS: macOS 11.2.3
 - VSCode version version : 1.76.1

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Additional context**
N/A

qarlosalberto, 2023-03-14:
What should be the correct command?

bit0fun, 2023-03-27:
Not sure if this was ever updated, but it appears from the GHDL plugin that the format should be :

> Alternatively, it is possible to analyze, elaborate and synthesize VHDL sources at once, instead of calling GHDL and Yosys in two steps. In this example:
> `yosys $MODULE -p 'ghdl leds.vhdl spin1.vhdl -e leds; synth_ice40 -json leds.json'`

With the mention of $MODULE being the following:

> ` NOTE: if GHDL is built as a module, set MODULE to '-m ghdl' or '-m path/to/ghdl.so', otherwise, unset it. `

It appears that currently, the module part is omitted.

bit0fun, 2023-03-27:
My best guess as to what is happening, is that [src/features/schematic.ts:201](https://github.com/TerosTechnology/vscode-terosHDL/blob/main/src/features/schematic.ts#L201) is not being set properly, as [src/features/schematic.ts:214](https://github.com/TerosTechnology/vscode-terosHDL/blob/main/src/features/schematic.ts#L214) appears to select the right thing for the plugin variable and then go on to be used on line [228](https://github.com/TerosTechnology/vscode-terosHDL/blob/main/src/features/schematic.ts#L228)

Though I may be mistaken if this is indeed a bug that is resolved or not, as I only appear to be running v2.0.2 as opposed to the latest available on GitHub. VSCode only appears to have v2.0.2 as the latest

qarlosalberto, 2023-03-27:
I will take a look. Currently I'm working in the version 3.x but fixing
bugs on 2.x

qarlosalberto, 2023-04-02:
I have updated the version in open-vsx