TerosTechnology/vscode-terosHDL #384: Documentation - process interconnection

AndreaCampera, 2023-03-23:
Hello all,

one feature that would be nice to have in the generated documentation is a block diagram with processes interconnection.
Sometimes this helps to understand the code, especialoly when you take cod e written by someone else
For instance every process could be treated like an entity, with input (what is read by the process) and outpus. Then connect the processes among them, and/or to the entity inputs/outputs.
This is more important than the schematic as it is now in my humbel opinion.
A selection of the nets/processes to be displayed would also be super important especially for big modules with a lot of processes

Thanks
Andrea

qarlosalberto, 2023-03-23:
Hi!

It's a very good idea. But really hard to implement. This would take a long time...

AndreaCampera, 2023-03-23:
I would like to help,

Actually we have a lot of scripts to parse the code and create structures with processes, inputs and outputs. That’s easy.
What we don’t have is the graphic capability, I was thinking as you already do the state machines (I don’t know what you are using, graphviz?)

Andrea

[Logo colori web]
Andrea Campera
Chief Executive Officer
Campera Electronic Systems Srl, www.campera-es.com<http://www.campera-es.com/>
Headquarter: Via Enrico Mayer 69, 57125, Livorno, Italy,
Mob:  +39 389 6426795  -  Tel:  +39 0586 371326
Email: ***@***.******@***.***> Pec: ***@***.***

This email and any attachments are confidential and is intended only for the use of the intended addressee thereof. If you are not the intended addressee, please be aware that any use, disclosure, distribution and/or copying of this communication, either whole or partial, is strictly prohibited. If you receive this communication in error, please notify the sender immediately and delete it from your computer

el3ctrician, 2025-02-08:
hi i could take this into had if we discuss these scripts together ...