TerosTechnology/vscode-terosHDL #392: Signal (WaveDrom) change font size in State machine viewer

StephaneIFO, 2023-04-19:
I discovered TerosHDL a few days ago and it is really a great tool, nice work!
But I think there is a conflict in the css rules between signal (WaveDrom) and State machine viewer

**Describe the bug**
When you add a signal (WaveDrom) the font size in the State machine viewer increase

**To Reproduce**
Add this `--! { signal: [{ name: "Alfa", wave: "01.zx=ud.23.456789" }] }` to the line 25 of state_machine_1.vhd  
https://github.com/TerosTechnology/teroshdl-examples/blob/main/state_machine/state_machine_1.vhd#L25
=> look at the font size in the state machine

**Please complete the following information:**
 - OS: Windows 10 
 - VSCode v1.77.3
 - TerosHDL v2.0.13

**Screenshots**
state_machine_1 without signal => OK
[image]

state_machine_1 with signal => NOK
[image]

StephaneIFO, 2023-08-21:
Hi, update with v5.0.9 : 

File https://github.com/TerosTechnology/teroshdl-examples/blob/main/state_machine/fsm_6.vhd
Add on line 1 : --! { signal: [{ name: "Alfa", wave: "01.zx=ud.23.456789" }] }

With signal : 
* Diagram font size reduce
* State machines font size increase

[image]

qarlosalberto, 2023-08-21:
Try with this version: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

StephaneIFO, 2024-08-23:
Hi, with 6.0.1(pre-release) it's worse : an fsm crashes the whole documentation
Without fsm : ok
[image]

With fsm : no documentation and error "Cannot find module 'fast-json-stable-stringify'"
`Cannot find module 'fast-json-stable-stringify'
Require stack:
- c:\Users\xxxxxx\.vscode\extensions\teros-technology.teroshdl-6.0.1\node_modules\ajv\lib\compile\index.js
- c:\Users\xxxxxx\.vscode\extensions\teros-technology.teroshdl-6.0.1\node_modules\ajv\lib\ajv.js
- c:\Users\xxxxxx\.vscode\extensions\teros-technology.teroshdl-6.0.1\node_modules\teroshdl2\node_modules\state-machine-cat\dist\commonjs\bundle.js
- c:\Users\xxxxxx\.vscode\extensions\teros-technology.teroshdl-6.0.1\node_modules\teroshdl2\node_modules\state-machine-cat\dist\commonjs\index.js
- c:\Users\xxxxxx\.vscode\extensions\teros-technology.teroshdl-6.0.1\node_modules\teroshdl2\out\parser\fsm_base_parser.js
- c:\Users\xxxxxx\.vscode\extensions\teros-technology.teroshdl-6.0.1\node_modules\teroshdl2\out\parser\ts_vhdl\fsm.js
- c:\Users\xxxxxx\.vscode\extensions\teros-technology.teroshdl-6.0.1\node_modules\teroshdl2\out\parser\ts_vhdl\parser.js
- c:\Users\xxxxxx\.vscode\extensions\teros-technology.teroshdl-6.0.1\node_modules\teroshdl2\out\parser\factory.js
- c:\Users\xxxxxx\.vscode\extensions\teros-technology.teroshdl-6.0.1\node_modules\teroshdl2\out\documenter\documenter.js
- c:\Users\xxxxxx\.vscode\extensions\teros-technology.teroshdl-6.0.1\node_modules\teroshdl2\out\documenter\export_t.js
- c:\Users\xxxxxx\.vscode\extensions\teros-technology.teroshdl-6.0.1\node_modules\teroshdl2\out\index.js
- c:\Users\xxxxxx\.vscode\extensions\teros-technology.teroshdl-6.0.1\out\features\language_provider\lsp\rust_hdl.js
- c:\Users\xxxxxx\.vscode\extensions\teros-technology.teroshdl-6.0.1\out\features\language_provider\language_provider.js
- c:\Users\xxxxxx\.vscode\extensions\teros-technology.teroshdl-6.0.1\out\teroshdl.js
- c:\Users\xxxxxx\.vscode\extensions\teros-technology.teroshdl-6.0.1\out\extension.js
- e:\Program Files\Microsoft VS Code\resources\app\out\bootstrap-fork.js`
[image]

qarlosalberto, 2024-08-26:
> Hi, with 6.0.1(pre-release) it's worse : an fsm crashes the whole documentation Without fsm : ok [...]

This version should fix it: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/v6.0.2beta
But not the problem with the css

StephaneIFO, 2024-08-27:
Hi, FSM work with 6.0.3, but as you say the css problem stay
TerosHDL is great but I need chronograms and FSM in my documentation, so, in all honesty, will this problem ever be solved or should I definitely stop hoping and move on to another solution for my VHDL documentation? 

[image]

qarlosalberto, 2024-08-27:
I don't know when I will fix the bug. If you need commercial support you can send me an email to: carlosruiznaranjo@gmail.com