TerosTechnology/vscode-terosHDL #395: Export state machine diagram to structured text

svenn71, 2023-04-24:
Sometimes I would like to open the resulting state diagram in a separate program to experiment with the presentation.
Currently the only option is "Save as SVG".
A JSON or YAML file would be fine, then further processing can be done in an external utility to adjust to target application as there are several graph description languages available.

qarlosalberto, 2023-04-24:
What could be the format?

svenn71, 2023-04-25:
My initial thought was to play around with the state machine diagram in [yEd ](https://www.yworks.com/products/yed#yed-support-resources) since yEd give me the possibility to play around with layout of the graph. yEd supports the following file formats https://yed.yworks.com/support/manual/fileformat.html

I think the simple [GML ASCII format](https://en.wikipedia.org/wiki/Graph_Modelling_Language) looks best as a starting point, because it supports a few other tools than yEd and also graphviz (through a converter) according to the linked Wikipedia page.

With GML there is also support for python module NetworkX which I see used in Jupyter from time to time.