TerosTechnology/vscode-terosHDL #397: Trouble activating virtual environment on Windows

tj-scherer, 2023-05-01:
Hello! I am experiencing an issue with using my virtual environment on Windows. I have created it directly through vscode in my workspace, and it is named `.venv`. Following the guide [here](https://terostechnology.github.io/terosHDLdoc/configuration/python.html#example-with-virtualenv) indicates that the python path is showing my python path to be the main installation, not the activated virtual environment. It also shows that the deps are not installed, which they are (on the venv). Is this still the correct thing to put in the path config?

`. .venv/bin/activate & python`

 - OS: Windows 10
 - VSCode version version 1.77.3

[image]
[image]
[image]

tafkos, 2023-08-03:
@tj-scherer Try to activate environment in command line first (cmd.exe), then run `code` from command line with activated environment. Works for `conda`.

tj-scherer, 2023-08-09:
I was able to make this work by creating a "global" virtual environment specifically for Teros.

```
cd C:/work/teroshdl
python -m venv .venv
.venv/Scripts/activate
pip install teroshdl
```

Then, my Teros configuration contains `C:/work/teroshdl/.venv/Scripts/python.exe` in the python3 binary path, no environment activation required.

I had the misconception that I should be pip installing teros in every workspace, but this approach makes much more sense overall. The new documentation definitely helped clear this up.

closed 2023-08-09 by tj-scherer