TerosTechnology/vscode-terosHDL #398: Wrapper/instance generator

jagjordi, 2023-05-05:
**Is your feature request related to a problem? Please describe.**
Automatically generate an instance of a given module/entity

**Describe the solution you'd like**
Similar how emacs vhdl-mode does it 

**Describe alternatives you've considered**
Some external scripts like `genwrapperverilog` but it would be better to have a tight integration

jagjordi, 2023-05-05:
For example I have a module: 
```sv
module my_module #(
    parameter WIDTH = 8,
    parameter DEPTH = 16
    ) (
    input clk,
    input rst,
    input [WIDTH-1:0] data,
    output [WIDTH-1:0] result
);
```

I invoke this "instance generator" and it generates:
```sv
my_module #(
    .WIDTH(8),
    .DEPTH(16)
    ) my_module_inst (
    .clk(clk),
    .rst(rst),
    .data(data),
    .result(result)
);
```

The instance generator could be invoked from a file, then a list of entities/modules in the project can be displayed and the user selects one of them to generate the instanciatino code

qarlosalberto, 2023-05-05:
This one? https://terostechnology.github.io/terosHDLdoc/templates/start.html

qarlosalberto, 2023-05-05:
De nada :)

closed 2023-05-06 by jagjordi