TerosTechnology/vscode-terosHDL #404: "Hover & evaluate“ feature doesn't support ".sv" files.

Lvwings, 2023-06-12:
Thanks.

qarlosalberto, 2023-06-12:
Can you provide a code sample?

Lvwings, 2023-06-12:
Thanks for your reply. "Hover & evaluate“ feature works great when I use .v files. The code sample shows below:

module code_sample (
    input  logic clk,     // Clock
    input  logic clk_en,  // Clock Enable
    input  logic rst_n,   // Asynchronous reset active low
    output logic out

);
    logic r_out = '0;
    always_ff @(posedge clk or negedge rst_n) begin : proc_
        if (~rst_n) begin
            out <= 0;
        end else if (clk_en) begin
            out <= 1;
        end else begin
            out <= r_out;
        end
    end

endmodule : code_sample