TerosTechnology/vscode-terosHDL #407: New user: adding files and directories

aspdigital, 2023-06-20:
I'm evaluating terosHDL with a current project. The project has a lot of source files, and the source tree is hierarchical. The source tree is also a Subversion working copy, and this makes adding source files to the project rather annoying, as "Load all files in folder and subfolders" is indiscrimate. That is, it adds _all_ files to the project, including the .svn folder in each of the source tree's subfolders. Obviously this is not what we want! 

A filter here would be very handy, something that lets the import tool ignore directories like the .svn folder.

There is no documentation that I can find for the CSV import option. Please provide documentation and a simple example!

qarlosalberto, 2023-06-20:
Could you send me an email to carlosruiznaranjo@gmail.com?

aspdigital, 2023-06-20:
Yes, will do.