TerosTechnology/vscode-terosHDL #409: I following terosHDLdoc's Graphical mode guide to do ,but failed,can you help me ,thank you very much

hmz2015011117, 2023-06-24:
****************************************************************************************************************************************
---> Documentation: https://terostechnology.github.io/terosHDLdoc/project_manager/configuration.html
---> Project name: project_ghdl
---> Top level: half_adder_process_tb    
---> Tool name: ghdl     
---> Installation path: System path     
---> Open tool GUI: true
****************************************************************************************************************************************
2023-06-24 21:18:45: C:\Users\hmz\AppData\Local\Programs\Python\Python39\python.exe "c:\Users\hmz\.vscode\extensions\teros-technology.teroshdl-2.0.13\resources\project_manager\run_edalize.py" "C:\Users\hmz\AppData\Local\Temp\edam.json" "ghdl" "" "gui" "false" "vcdrom"
"Creating libraries directories"
ghdl -i --std=08 --ieee=synopsys  c:\Users\hmz\Desktop\1\half_adder_tb.vhd c:\Users\hmz\Desktop\1\half_adder.vhd
process_begin: CreateProcess(NULL, ghdl -i --std=08 --ieee=synopsys c:\Users\hmz\Desktop\1\half_adder_tb.vhd c:\Users\hmz\Desktop\1\half_adder.vhd, ...) failed.
make (e=2): ϵͳ�Ҳ���ָ�����ļ���

make: *** [work-obj08.cf] ���� 2

************************************************************************************************
---> Build directory: C:\Users\hmz\.teroshdl\build
C:\Users\hmz\AppData\Local\Temp\edam.json
---> The test has finished with errors!
****************************************************************************************************************************************

[image]

hmz2015011117, 2023-06-24:
I Know , i forgot download ghdl,but which version can i download?
[image]

qarlosalberto, 2023-06-29:
If you want to install GHDL, follow: https://www.drpi.fr/en/blog/2021-12-28-ghdl-on-windows.html

closed 2023-07-29 by qarlosalberto