TerosTechnology/vscode-terosHDL #410: support gdsfactory

joamatab, 2023-07-04:
What do you think of adding [gdsfactory](https://gdsfactory.github.io/gdsfactory/) support to Teros?

gdsfactory could expand teros into analog/photonics/MEMS design

gdsfactory supports also different simulators that could be included

MVP:

- [ ] include gdsfactory new template project `cookiecutter https://github.com/joamatab/cookiecutter-pypackage-minimal`
- [ ] include filewacher support `gf watch folderName`
- [ ] detect `.pic.yml` syntax with autocomplete [see](https://github.com/gdsfactory/vscode-gdsfactory)

qarlosalberto, 2023-07-04:
It's a good idea. But I prefer to add all the changes related to gdsfactory after the next release (this/next week). I'm done fixing the bugs and I don't want to add new ones

qarlosalberto, 2025-03-18:
In a near? future :D

closed 2025-03-18 by qarlosalberto