TerosTechnology/vscode-terosHDL #414: Finite State Machine (FSM) breaks Documentation with error Cannot find module 'fast-json-stable-stringify'

Luca-s-Works, 2023-07-05:
**Describe the bug**
Hello, just upgraded from v2.0.13 to v5.0.0 via manual VSIX download from this Git releases (VSC plug-in manager cannot find anything past v2.0.13).

After v5 installation and VSC relaunch, I am experiencing issues with automatic documentation extraction feature **regarding ALL VHDL sources that implement a FMS within**.

This error occurs whenever I ask Teros to extract documentation for a VHDL source with a FSM in it:
```
Cannot find module 'fast-json-stable-stringify'
Require stack:
- ~/.vscode-server/extensions/teros-technology.teroshdl-5.0.0/node_modules/ajv/lib/compile/index.js
- ~/.vscode-server/extensions/teros-technology.teroshdl-5.0.0/node_modules/ajv/lib/ajv.js
- ~/.vscode-server/extensions/teros-technology.teroshdl-5.0.0/node_modules/teroshdl2/node_modules/state-machine-cat/dist/commonjs/bundle.js
- ~/.vscode-server/extensions/teros-technology.teroshdl-5.0.0/node_modules/teroshdl2/node_modules/state-machine-cat/dist/commonjs/index.js
- ~/.vscode-server/extensions/teros-technology.teroshdl-5.0.0/node_modules/teroshdl2/out/parser/fsm_base_parser.js
- ~/.vscode-server/extensions/teros-technology.teroshdl-5.0.0/node_modules/teroshdl2/out/parser/ts_vhdl/fsm.js
- ~/.vscode-server/extensions/teros-technology.teroshdl-5.0.0/node_modules/teroshdl2/out/parser/ts_vhdl/parser.js
- ~/.vscode-server/extensions/teros-technology.teroshdl-5.0.0/node_modules/teroshdl2/out/parser/factory.js
- ~/.vscode-server/extensions/teros-technology.teroshdl-5.0.0/node_modules/teroshdl2/out/documenter/documenter.js
- ~/.vscode-server/extensions/teros-technology.teroshdl-5.0.0/node_modules/teroshdl2/out/documenter/export_t.js
- ~/.vscode-server/extensions/teros-technology.teroshdl-5.0.0/node_modules/teroshdl2/out/index.js
- ~/.vscode-server/extensions/teros-technology.teroshdl-5.0.0/out/features/language_provider/lsp/rust_hdl.js
- ~/.vscode-server/extensions/teros-technology.teroshdl-5.0.0/out/features/language_provider/language_provider.js
- ~/.vscode-server/extensions/teros-technology.teroshdl-5.0.0/out/teroshdl.js
- ~/.vscode-server/extensions/teros-technology.teroshdl-5.0.0/out/extension.js
- ~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/loader.js
- ~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/bootstrap-amd.js
- ~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/bootstrap-fork.js
```
I cannot npm install on the remote server of my company (where the sources resides) since I have not enough user privileges.

Documentation is created flawlessly for VHDL sources with no FSM in it. **The code section that is causing the issue is the process within the architecture section of the state machine**. This makes me think the issue is actually more related to the FSM Extract feature.
**Though, asking Teros to draw the diagram with FSM Extract produce no result, yet the above mentioned exception is not thrown.**

**To Reproduce**
I was able to reproduce this problem simply with the [state machine VHDL examples in teroshdl documentation](https://github.com/TerosTechnology/teroshdl-examples/tree/main/state_machine).

**Code**
Code to reproduce the error: [TerosHDL examples for State Machine documentation](https://github.com/TerosTechnology/teroshdl-examples/tree/main/state_machine)

**Please complete the following information:**
VSCode version 1.79.2 (system setup)
Commit: 695af097c7bd098fbf017ce3ac85e09bbc5dda06
Date: 2023-06-14T08:57:04.379Z
Electron: 22.5.7
Chromium: 108.0.5359.215
Node.js: 16.17.1
V8: 10.8.168.25-electron.0
OS: Windows_NT x64 10.0.19045

**Additional context**
I am using Remote VSC extension to access Linux work server where the sources are.

qarlosalberto, 2023-07-05:
Use: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/5.0.1

Luca-s-Works, 2023-07-05:
> Use: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/5.0.1

This VSIX stalls and installation never completes. Tried also relaunching, uninstalling previous version of the Teros extension

Luca-s-Works, 2023-07-05:
> And latest? https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

Same, hangs on:
```
$ ./code-server --install-extension "~/2023-07-05__09_00_teroshdl-5.0.1.vsix"
Installing extensions...
```

Luca-s-Works, 2023-07-05:
Set log level to trace and this is the result of the --install-extension command:
```
[12:56:57] [127.0.0.1][7a3194fa][ManagementConnection] The client has reconnected.
[12:56:57] [127.0.0.1][01a6df0d][ExtensionHostConnection] The client has reconnected.
rejected promise not handled within 1 second: CodeExpectedError: Could not find pty on pty host
stack trace: CodeExpectedError: Could not find pty on pty host
    at k.L (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:28:6920)
    at k.updateIcon (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:28:2677)
    at Object.call (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:17:5058)
    at h.s (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:15:5412)
    at h.q (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:15:4915)
    at ~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:15:4326
    at l.invoke (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:11:145)
    at u.deliver (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:11:2125)
    at s.fire (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:11:1733)
    at process.z (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:9:21144)
    at process.emit (node:events:513:28)
    at emit (node:internal/child_process:939:14)
    at processTicksAndRejections (node:internal/process/task_queues:84:21)
[12:56:58] Error [CodeExpectedError]: Could not find pty on pty host
    at k.L (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:28:6920)
    at k.updateIcon (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:28:2677)
    at Object.call (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:17:5058)
    at h.s (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:15:5412)
    at h.q (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:15:4915)
    at ~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:15:4326
    at l.invoke (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:11:145)
    at u.deliver (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:11:2125)
    at s.fire (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:11:1733)
    at process.z (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:9:21144)
    at process.emit (node:events:513:28)
    at emit (node:internal/child_process:939:14)
    at processTicksAndRejections (node:internal/process/task_queues:84:21)
rejected promise not handled within 1 second: CodeExpectedError: Could not find pty on pty host
stack trace: CodeExpectedError: Could not find pty on pty host
    at k.L (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:28:6920)
    at k.updateTitle (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:28:2627)
    at Object.call (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:17:5058)
    at h.s (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:15:5412)
    at h.q (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:15:4915)
    at ~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:15:4326
    at l.invoke (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:11:145)
    at u.deliver (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:11:2125)
    at s.fire (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:11:1733)
    at process.z (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:9:21144)
    at process.emit (node:events:513:28)
    at emit (node:internal/child_process:939:14)
    at processTicksAndRejections (node:internal/process/task_queues:84:21)
[12:56:58] Error [CodeExpectedError]: Could not find pty on pty host
    at k.L (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:28:6920)
    at k.updateTitle (~/.vscode-server/bin/695af097c7bd098fbf017ce3ac85e09bbc5dda06/out/vs/platform/terminal/node/ptyHostMain.js:28:2627)
[... cut ...]
[IPC Library: Pty Host] DEBUG ChildProcessMonitor: Has child processes changed false
[IPC Library: Pty Host] TRACE IPty#onData Installing extensions...

[12:59:50] [File Watcher (node.js)] [raw] ["rename"] .23b86147-a44d-4bdf-befb-a6b6a646d3b8
[12:59:50] [File Watcher (node.js)] [ADDED] ~/.vscode-server/extensions/.23b86147-a44d-4bdf-befb-a6b6a646d3b8
[12:59:50] [File Watcher (node.js)] >> normalized [ADDED] ~/.vscode-server/extensions/.23b86147-a44d-4bdf-befb-a6b6a646d3b8
[IPC Library: Pty Host] TRACE IPty#pid
```

qarlosalberto, 2023-07-05:
I don't see any error related to TerosHDL. Can you try to install it from the GUI?

Luca-s-Works, 2023-07-05:
Locally, I have no problem istalling the 5.0.1 pre-release VSIX.

On the remote VSC Server however, either from the GUI passing the VSIX locally or as on-server path, from GUI on extension column to install on remote server, or via cli on server it hangs:
```
[13:24:38] Started scanning system extensions
[13:24:38] Started scanning user extensions r {
  scheme: 'file',
  authority: '',
  path: '~/.vscode-server/extensions/extensions.json',
  query: '',
  fragment: '',
  _formatted: 'file://~/.vscode-server/extensions/extensions.json',
  _fsPath: '~/.vscode-server/extensions/extensions.json'
}
[13:24:38] Scanned user extensions: 4
[13:24:38] Scanned system extensions: 30
[13:24:38] ExtensionGalleryService#download teros-technology.teroshdl
[13:24:38] resolveShellEnv(): running (macOS/Linux)
[13:24:49] ExtensionGalleryService#downloadSignatureArchive teros-technology.teroshdl
[13:24:49] resolveShellEnv(): running (macOS/Linux)
[13:24:52] Extension signature is verified: teros-technology.teroshdl
[13:24:52] Extracting extension... teros-technology.teroshdl
[13:24:52] Started extracting the extension from ~/.vscode-server/data/CachedExtensionVSIXs/teros-technology.teroshdl-2.0.13 to ~/.vscode-server/extensions/.a62e4cf7-0af0-4777-82b3-6d4dc283028f
[13:24:53] [File Watcher (node.js)] [raw] ["rename"] .a62e4cf7-0af0-4777-82b3-6d4dc283028f
[13:24:53] [File Watcher (node.js)] [ADDED] ~/.vscode-server/extensions/.a62e4cf7-0af0-4777-82b3-6d4dc283028f
[13:24:53] [File Watcher (node.js)] >> normalized [ADDED] ~/.vscode-server/extensions/.a62e4cf7-0af0-4777-82b3-6d4dc283028f
New EH opened, aborting shutdown
[13:28:50] New EH opened, aborting shutdown
```

qarlosalberto, 2023-07-05:
I see: ~/.vscode-server/data/CachedExtensionVSIXs/teros-
technology.teroshdl-2.0.13

Can you remove that folder?

qarlosalberto, 2023-07-05:
https://github.com/microsoft/vscode-remote-release/issues/1665

I think that the issue it's not related to TerosHDL.

Luca-s-Works, 2023-07-05:
> I see: ~/.vscode-server/data/CachedExtensionVSIXs/teros- technology.teroshdl-2.0.13 Can you remove that folder? 

I've removed, I've also tried starting clean removing the whole ~/.vscode-server. I see in the log though:
```
 ExtensionGalleryService#download teros-technology.teroshdl
```

> [microsoft/vscode-remote-release#1665](https://github.com/microsoft/vscode-remote-release/issues/1665)
> 
> I think that the issue it's not related to TerosHDL.

I've read that. It's true, v5.0.0 installs though (either via GUI or CLI)"
```
Installing extension: teros-technology.teroshdl
[15:37:49] Extracted extension to ~/.vscode-server/extensions/.f84c818d-c2a3-4796-9e2f-235ef1e038d9: teros-technology.teroshdl
[15:37:49] Renamed to ~/.vscode-server/extensions/teros-technology.teroshdl-5.0.0
[15:37:49] Extracting extension completed. teros-technology.teroshdl
[15:37:49] Extension installed successfully: teros-technology.teroshdl
```

qarlosalberto, 2023-07-29:
Please, use the latest version 5.0.7. Feel free to re-open the issue if the problem continues.

closed 2023-07-29 by qarlosalberto