TerosTechnology/vscode-terosHDL #415: Difference between v2 and v5

wimille, 2023-07-05:
Hi,

maybe i don't see the associated post or message, but what is the difference between version v2.0.13 available on the vscode market and version 5.0.1 available throught the github repository ?

Sorry if the question has been already asked ?

Regards

qarlosalberto, 2023-07-05:
- I have fixed a lot of bugs.
- Support for Raptor in RapidSilicon.
- Watchers for VUnit and CSV.
- New tool manager.
- New configuration page.

https://terostechnology.github.io/teroshdl-docusaurus-doc/

wimille, 2023-07-06:
Ok thank you for your answer. 
Why this version is not available through vscode marketplace ?

qarlosalberto, 2023-07-06:
Because it's a beta version.

wimille, 2023-07-06:
Thank you for all this precisions. 
I really enjoy your work.

closed 2023-07-06 by wimille