TerosTechnology/vscode-terosHDL #417: Formatter set to Verible, but removes newline from end of file

michael-etzkorn, 2023-07-08:
**Describe the bug**
A clear and concise description of what the bug is.

With verible-verilog-format installed, run the formatter. 

**To Reproduce**
Code to reproduce the error.
```
module test;
endmodule

```
Run format on the above with a newline and config set to Verible. Running this from the CLI will not change the newline, but running from within vscode with TerosHDL removes the newline.
**Code**
Code to reproduce the error.

**Please complete the following information:**
 - OS: [e.g.Ubuntu 18.04]
 - VSCode version version [e.g. 22]
CentOS7
VSCODE: 1.74.3

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

This occurs even if verible would fail formatting, so I think the issue is TerosHDL doing some additional format operation when I run terosHDL.format

michael-etzkorn, 2023-07-08:
This did not previously happen with TerosHDL. I have v5.0.1 installed

qarlosalberto, 2023-07-08:
What is your veriboe config?

qarlosalberto, 2023-07-10:
fixed: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

closed 2023-07-10 by qarlosalberto