TerosTechnology/vscode-terosHDL #418: The Vivado linter still uses Verilog syntax for checking SystemVerilog.

hhucchenyixiao, 2023-07-13:
[image]
Can I make the linter work properly by adding any parameters?

SebastianHambura, 2023-07-13:
Same problem here after the upgrade to the v5.0. 
Inspired by this : https://support.xilinx.com/s/question/0D52E00006hpQq2SAE/xvlog-command-error-for-sv-files?language=en_US I've tried adding '-sv' but that didn't have any impact. 
Would also love for this to work again :)

SebastianHambura, 2023-07-14:
Did a quick test, seems to be working :)
Thanks for being so quick !

closed 2023-07-14 by qarlosalberto

hhucchenyixiao, 2023-07-14:
> Try this version: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

Thank you, the bug fix was very effective.