TerosTechnology/vscode-terosHDL #419: The highlight for `ifdef `else `endif `ifndef is missing

beanspower, 2023-07-14:
Hello expert,
**Describe the bug**
The hightlight for " ifdef" "else" "endif“ ”ifndef“ are missing.

**Please complete the following information:**
 - OS: Win 10
 - VSCode version version 1.80.1
 - Teros HDL : v5.0.2
[image]

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

I try to modify the "verilog.configuration.json" , but it seems not  good enough.
Add below codes in brackets
		["`ifdef","`endif"],
		["`ifndef","`endif"],
[image]

beanspower, 2023-07-18:
@qarlosalberto Hello Expert ， could you help to check it？

qarlosalberto, 2023-07-18:
Please share the code.

beanspower, 2023-07-18:
@qarlosalberto Hello expert , 
I share one example as the code in https://github.com/TerosTechnology/vscode-terosHDL/issues/296
Thanks for your kind help.
[Example.zip](https://github.com/TerosTechnology/vscode-terosHDL/files/12082019/Example.zip)