TerosTechnology/vscode-terosHDL #420: If path has white space, linter won't work

Devil-SX, 2023-07-16:
**Describe the bug**
We can see that teroshdl used both double quotes and escape characters, resulting in the inability to handle spaces correctly.

**Please complete the following information:**
 - Ubuntu22.04
 - VSCode version 1.79.2

**Screenshots**
[image]

qarlosalberto, 2023-07-17:
Check if the latest version in the market (5.0.5) fixes it.

elpollodiablo84, 2023-07-18:
If I'm not mistaken, it seems there is the same problem in the dependency graph generator, since I'm getting an error stating that "C:\Program" is not a known command (vscode is installed in "C:\Program Files\....").

qarlosalberto, 2023-07-18:
Could you show a screenshot with the error?

elpollodiablo84, 2023-07-18:
[image]
The error is in italian maybe because is a system error?

qarlosalberto, 2023-07-18:
Try this version: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

Not sure if your problem is related to the VSCode installation path or due to the Python path. Has your Python path white spaces?

elpollodiablo84, 2023-07-18:
Both my python path and vscode path have spaces. I tried your suggested version but the error is still there...

qarlosalberto, 2023-07-18:
Try with this version: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

elpollodiablo84, 2023-07-18:
> Try with this version: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

I still got the error...

> Try to escape you Python path and let me know if it works.

How should escape it? I'm using the system path, so I'm not declaring it in the configuration menu.
If the problem is the python path, why VSG is still working?

elpollodiablo84, 2023-07-19:
[image]
It seems its all fine

qarlosalberto, 2023-07-19:
I think that the problem is related to the white space in the Python path. This version should fix it: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

elpollodiablo84, 2023-07-19:
> I think that the problem is related to the white space in the Python path. This version should fix it: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

It works! 🎉🎉🎉
Well done, thank you!

qarlosalberto, 2023-07-29:
Please, use the latest version 5.0.7. Feel free to re-open the issue if the problem continues.

closed 2023-07-29 by qarlosalberto