TerosTechnology/vscode-terosHDL #422: cocotb not found, Make not found

Atomillo, 2023-07-17:
Hello!

I'm a really new user and have tried to follow all the installation steps, dowloading Python 3, doing the pip install, installing Cygwin, etc...
After all this however, when I click on "Check dependencies", I get the following log:

2023-07-17 15:28:40.561 [info] Checking dependencies...
2023-07-17 15:28:40.719 [info] --->  Python found: C:\Users\carcosbo\AppData\Local\Programs\Python\Python311\python.exe
2023-07-17 15:28:40.899 [info] --->  vunit found
2023-07-17 15:28:40.961 [error] --->  cocotb not found
2023-07-17 15:28:41.023 [info] --->  edalize found
2023-07-17 15:28:41.042 [error] ---> Make not found

Is there anywhere I can check what happened?

Many thanks!
Cai

qarlosalberto, 2023-07-17:
Have you installed `make`? https://terostechnology.github.io/terosHDLdoc/docs/getting_started/installation#make

```
3. Verify that Make has been installed by running the following command: make -v. This should display the version of Make installed on your system.
```

Atomillo, 2023-07-17:
Hi! 

Yes, when in Cygwin terminal I type make -v I get:
GNU Make 4.4.1
Built for x86_64-pc-cygwin
Copyright (C) 1988-2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

qarlosalberto, 2023-07-17:
Run again the help button with the new version of the market and share the output.

Atomillo, 2023-07-18:
When yo usay "help" you mean checking dependencies again after reloading? I get this:
2023-07-18 08:21:56.142 [info] Checking dependencies...
2023-07-18 08:21:56.294 [info] --->  Python found: C:\Users\carcosbo\AppData\Local\Programs\Python\Python311\python.exe
2023-07-18 08:21:56.486 [info] --->  vunit found
2023-07-18 08:21:56.549 [error] --->  cocotb not found
2023-07-18 08:21:56.611 [info] --->  edalize found
2023-07-18 08:21:56.632 [error] ---> Make not found
2023-07-18 08:21:56.632 [error] 'make' is not recognized as an internal or external command,
operable program or batch file.
2023-07-18 08:21:56.632 [error]

qarlosalberto, 2023-07-18:
> I was running into the same issue but fixed it. I installed make using
> Chocolatey following this site
> https://pakstech.com/blog/make-windows/#google_vignette and that seemed
> to work for everything except cocotb but I'm not sure how important that
> dependency is

Cocotb is optional.

Atomillo, 2023-07-18:
Using Chocolatey worked!
Thanks qarlosaberto and Kyle-912 for all your help!
After more than a year using just Vivado, seeing code autocomplete feels like dark magic

qarlosalberto, 2023-07-29:
I have added the possibility to configure `make` installation directory and `Chocolatey` installation to the doc.

closed 2023-07-29 by qarlosalberto