TerosTechnology/vscode-terosHDL #424: In FSM viewer, the transfer relationship is missing for case statement

beanspower, 2023-07-21:
Version: 1.80.1 (system setup)
Commit: 74f6148eb9ea00507ec113ec51c489d6ffb4b771
Date: 2023-07-12T17:22:07.651Z
Electron: 22.3.14
ElectronBuildId: 21893604
Chromium: 108.0.5359.215
Node.js: 16.17.1
V8: 10.8.168.25-electron.0
OS: Windows_NT x64 10.0.19045
TerosHDL：v5.0.5

When i use below code to generate the state machine viewer, 
```
module test (
    input clk,
    input rstn,
    input sel,
    output [5:0] out
  );

  reg [2:0]st_curr;
  reg [2:0]st_next;

  always @(posedge clk or negedge rstn)
  begin
    if(!rstn)
    begin
      st_curr <= 3'd0;
    end
    else
    begin
      st_curr = st_next;
    end
  end

  always @(*)
  begin
    case (st_curr)
      3'd0:
        st_next = 3'd1;
      3'd1:
        if(sel)
          st_next = 3'd2;
        else
          st_next = 3'd4;
      3'd2,3'd3:
        st_next = 3'd0;
      3'd4:
        st_next = 3'd3;
      default:
        st_next = 3'd0;
    endcase
  end
endmodule
```
The state transfer is missing for 3'd3.
But it is actually described in the verilog. 
```
      3'd2,3'd3:
        st_next = 3'd0;
```
[image]

beanspower, 2023-07-26:
@qarlosalberto
Hello expert,
Could you help to check this issue here? 
Thank for your kind help.

qarlosalberto, 2023-07-29:
I will take a look soon.