TerosTechnology/vscode-terosHDL #426: Requesting support for neovim

bentomo, 2023-07-21:
**Is your feature request related to a problem? Please describe.**
No issues related

**Describe the solution you'd like**
It would be excellent if some of the key features such as instantiation template generation and module documentation generation could be used in other editors like neovim

**Describe alternatives you've considered**
I am trial running using neovim as my main editor instead of VS code but I definitely miss TerosHDL. The plugins for neovim related to HDL are very limited.

**Additional context**
TerosHDL is easily the best and most complete HDL IDE that I've ever seen open, or closed. Supporting other editors, or at least a few of the best functions of TerosHDL, in other editors would increase its value and potentially get more contributors from other communities

I think the scope would be very limited to start. Just giving lua API access to some of the underlying tools and instructions on setup would be extremely useful. 

If I were to try to add support myself, would I need to fork this repository since it's labeled as the "vscode" repo? I would prefer if I could contribute upstream to the main repo.

qarlosalberto, 2023-07-21:
Could you send me an email? carlosruiznaranjo@gmail.com

bentomo, 2023-07-21:
Carlos and I spoke via email and he suggested that colibri may be a suitable candidate for this.

[https://github.com/TerosTechnology/vscode-terosHDL/tree/dev/packages/colibri](url)

I agree with him that this may be the path of least resistance for a neovim plugin, and that adding neovim support for this would take a great deal of effort.

bentomo, 2023-07-26:
When attempting to run the colibri from the command line I get node errors. Is there a list of what versions of node and NPM packages I should use? This is on a RHEL 8.4 system and I tried with Node 9, 10, and 12. I tried to follow ths instructions on the colibri page to do an `npm install` and `npm compile`

```
./bin/run teroshdl --help
  config reading core plugin /home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri +0ms
  config loadJSON /home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri/package.json +0ms
  config loadJSON /home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri/oclif.manifest.json +0ms
  config loadJSON /home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri/.oclif.manifest.json +0ms
  config:colibri2 loading IDs from /home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri/out/commands +0ms
  config:colibri2 found commands [ 'teroshdl:formatter', 'teroshdl:linter', 'teroshdl:template' ] +3ms
  config:colibri2 (require) /home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri/out/commands/teroshdl/formatter.js +38ms
  config:colibri2 (require) /home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri/out/commands/teroshdl/template.js +0ms
  config reading user plugins pjson /home/abcarter/.local/share/colibri2/package.json +0ms
  config loadJSON /home/abcarter/.local/share/colibri2/package.json +0ms
(node:317348) [ENOENT] Error Plugin: colibri2: ENOENT: no such file or directory, open '/home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri/out/reporter/html
_detailed_formatter/templates/main-page.html'
module: @oclif/core@1.26.2
task: toCached
plugin: colibri2
root: /home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri
See more details with DEBUG=*
(Use `node --trace-warnings ...` to show where the warning was created)
Error Plugin: colibri2: ENOENT: no such file or directory, open '/home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri/out/reporter/html_detailed_formatter/tem
plates/main-page.html'
    at Object.openSync (node:fs:603:3)
    at Object.readFileSync (node:fs:471:35)
    at Object.<anonymous> (/home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri/out/reporter/html_detailed_formatter/template-generator.js:12:222)
    at Module._compile (node:internal/modules/cjs/loader:1256:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1310:10)
    at Module.load (node:internal/modules/cjs/loader:1119:32)
    at Module._load (node:internal/modules/cjs/loader:960:12)
    at Module.require (node:internal/modules/cjs/loader:1143:19)
    at require (node:internal/modules/cjs/helpers:110:18)
    at Object.<anonymous> (/home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri/out/reporter/html_detailed_formatter/detailed.js:6:27)
module: @oclif/core@1.26.2
task: toCached
plugin: colibri2
root: /home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri
See more details with DEBUG=*
  config loading plugins [ '@oclif/plugin-help', '@oclif/plugin-plugins' ] +1ms
  config reading core plugin /home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri/node_modules/@oclif/plugin-help +0ms
  config loadJSON /home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri/node_modules/@oclif/plugin-help/package.json +0ms
  config reading core plugin /home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri/node_modules/@oclif/plugin-plugins +0ms
  config loadJSON /home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri/node_modules/@oclif/plugin-plugins/package.json +0ms
  config loadJSON /home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri/node_modules/@oclif/plugin-help/oclif.manifest.json +0ms
  config loadJSON /home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri/node_modules/@oclif/plugin-plugins/oclif.manifest.json +0ms
  config:@oclif/plugin-help using manifest from /home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri/node_modules/@oclif/plugin-help/oclif.manifest.json +0ms
  config:@oclif/plugin-plugins using manifest from /home/abcarter/work/side_projects/vscode-terosHDL/packages/colibri/node_modules/@oclif/plugin-plugins/oclif.manifest.json
 +0ms
  config config done +1ms
  config start init hook +0ms
  config init hook done +0ms
Check errors in HDL files.
```

qarlosalberto, 2023-07-30:
The command line interface it's not full functional. Feel free to do a PR

bentomo, 2023-07-30:
Ah, I think I was misunderstanding colibri's purpose. I see now in the docs that it's a typescript LIBRARY. That means that if I would like to use functionality similar to how VSCode does things, I need to write my own TS application that utilizes the library. Am I on the right track? 

I don't know much about typescript, other than it's a superset of javascript, which is typically run in web browsers. Not that I know much about javascript either  😀

qarlosalberto, 2023-07-30:
Yes, you are right

qarlosalberto, 2025-03-18:
I haven't time to do it. Sorry.

closed 2025-03-18 by qarlosalberto