TerosTechnology/vscode-terosHDL #431: Verible not working in newest version (v5.0.6)

SebastianHambura, 2023-07-25:
**Describe the bug**
I'm using verible for formatter and style checking of Verilog/systemverilog files. In the newest version (5.0.6) it doesn't work : ```[info] Error format code.```. If I roll back to a previous version (e.g. 5.0.5), without changing anything in the file or my settings, it works again. 

**To Reproduce**
Update to v5.0.6 and try to format the following code. (Bonus : name the file something else, and you'll see you don't get a warning from verible for the module-file name missmatch)

**Code**
```
module test (
    input clk,
    input reset,
    input [31:0]   a,
    input [31:0] b,
    output [31:0] c
);

endmodule
````

**Please complete the following information:**
 - OS: Ubuntu 18.04.6
 - VSCode version version : 1.80.1 (I think)

**Screenshots**
N/A

**Additional context**
Rolling back to v5.0.5 fixes the issue for me, so I'll do that for now.

qarlosalberto, 2023-07-25:
What is your verible path?

SebastianHambura, 2023-07-25:
[image]

```~/Desktop/Tools/verible/bin```

qarlosalberto, 2023-07-29:
Fixed in the new version (5.0.7). Feel free to re-open the issue if the problem continues.

closed 2023-07-29 by qarlosalberto