TerosTechnology/vscode-terosHDL #432: '""' not recognized as an internal or external command

pintert3, 2023-07-26:
I keep getting the error: `'""' not recognized as an internal or external command` when I try to run, ever since I updated to v5. When I click `Check dependencies` it says:

```
2023-07-26 08:01:11.688 [error] ---> Python not found
2023-07-26 08:01:12.073 [info] --->  Make found
```

 - OS: Windows 11
 - VSCode version version 1.80.1

qarlosalberto, 2023-07-26:
What is your python3 configuration path?

qarlosalberto, 2023-07-26:
And when you have the error?

pintert3, 2023-07-26:
I'm using the Windows default path. And the error happened yesterday, almost immediately after updating terosHDL

qarlosalberto, 2023-07-26:
Could you share a screenshot of the error?

pintert3, 2023-07-26:
[image]

There you go!

pintert3, 2023-07-27:
It appears it's cause it couldn't figure out the python installation. I now have to manually insert it in the configuration menu in `General` for it to work. It now works when I input `C:/Users/Asus/AppData/Local/Programs/Python/Python311/python.exe` as the install path.

qarlosalberto, 2023-07-27:
@dsioffe you need to install Vunit: https://terostechnology.github.io/terosHDLdoc/docs/getting_started/installation#python3-package-dependencies

dsioffe, 2023-07-28:
Thanks!I did it.Now there is one error left:cocotb not foundHow to fix it?

qarlosalberto, 2023-07-28:
Cocotb it's an optional dependency

qarlosalberto, 2023-07-29:
I have added more log in 5.0.7

closed 2023-07-29 by qarlosalberto

pintert3, 2023-07-29:
Meanwhile, does this mean no more auto detecting the default python installation on windows? Cause the documentation says it should be able to.

qarlosalberto, 2023-07-29:
It does. But I don't know what is happening in your system.