TerosTechnology/vscode-terosHDL #434: Keep default config and use relative paths of current directory

michael-etzkorn, 2023-07-28:
I have two issues with verilator currently on v5

- My verilator config isn't saved between launches of vscode over ssh. 
  - The default config shouldn't disappear between launches of vscode server
- verilator relative arguments should work with the config. If I pass `-Irtl`, TerosHDL seems to use a different workspace

I'm currently using vscode ssh on windows to a server on CentOS7.

qarlosalberto, 2023-07-28:
Can you pass a screenshot with your Verilator config?

michael-etzkorn, 2023-07-28:
I don't think it's necessary to have my config to reproduce the issue with the current config. 

If you change any settings and close out of the config menu, the config can be lost. 
I'm not sure how easy that is to debug or reproduce, so it'd maybe just be easier to let users set a default config location in `settings.json` in vscode.

qarlosalberto, 2023-07-28:
I'm talking about your second issue. Can you explain it more detailed?

michael-etzkorn, 2023-07-28:
I can open a new issue for that. 

Say I have the following directory structure:
```
-> workspace_top
   top.v 
   -> rtl 
      rtl.v 
```
top.v dummy
```verilog

module top ( 
  input clk, 
  input rst
); 
  rtl u_rtl ( 
   .clk(clk),  
   .rst(rst)
  );  
endmodule 
```

rtl.v dummy

```verilog
module rtl ( 
  input rst,
  input clk
); 
endmodule
```

workspace is opened with: 
`code .` while in the workspace folder. 

When I pass `-Irtl` into verilator, I still get that rtl.v is not found. 

On another setup, I have seen relative paths work, but it's pointing to a completely different workspace that just happens to have the same RTL.

michael-etzkorn, 2023-07-28:
I've moved the relative paths bug to a separate issue #435. This issue is a feature request for a default config setting (similar to how the old major version of TerosHDL handled things).

qarlosalberto, 2023-07-28:
What do you mean with "default config setting"? The approach should be the same in the old TerosHDL and in v5

michael-etzkorn, 2023-07-28:
old TerosHDL would save config settings permanently. v5 doesn't save config settings if I close out of TerosHDL configuration.

michael-etzkorn, 2023-07-28:
There's also no longer a way to open the config settings directly with a developer command like there was in the old TerosHDL.

Currently just have "show TerosHDL -> Open Configuration Menu" but I have to make sure I export before closing the configuration menu because the settings disappear the next time the configuration menu is opened

[image]

qarlosalberto, 2023-07-28:
Hace you the same problem in local or only over SSH?

Does it happen with all the config?

What is your TerosHDL version?

qarlosalberto, 2023-07-28:
Let me know what field it's not saved.

michael-etzkorn, 2023-10-24:
This problem seems fixed as of `v5.0.12`

closed 2023-10-24 by michael-etzkorn