TerosTechnology/vscode-terosHDL #435: Relative Paths fail for verilog in certain workspace conditions

michael-etzkorn, 2023-07-28:
Say I have the following directory structure:
```
-> workspace_top
   top.v 
   -> rtl 
      rtl.v 
```
top.v dummy
```verilog

module top ( 
  input clk, 
  input rst
); 
  rtl u_rtl ( 
   .clk(clk),  
   .rst(rst)
  );  
endmodule 
```

rtl.v dummy

```verilog
module rtl ( 
  input rst,
  input clk
); 
endmodule
```

workspace is opened with: 
`code .` while in the workspace folder. 

When I pass `-Irtl` into verilator, I still get that rtl module is not found. 

On another setup, I have seen relative paths work, but it's pointing to a completely different workspace that just happens to have the same RTL.

_Originally posted by @michael-etzkorn in https://github.com/TerosTechnology/vscode-terosHDL/issues/434#issuecomment-1655229243_

michael-etzkorn, 2023-07-28:
[image]

[image]

[image]

Verilator will only find `rtl/rtl.v` if I change `-Irtl` to an absolute path. I have seen relative paths work, but they seem to point to another workspace.

michael-etzkorn, 2023-07-28:
related issue: #284 

It seems TerosHDL is somehow remembering the first workspace it was installed with or opened on and not refreshing workspace directory.

qarlosalberto, 2023-07-29:
Fixed in the new version (5.0.7). Feel free to re-open the issue if the problem continues.

closed 2023-07-29 by qarlosalberto