TerosTechnology/vscode-terosHDL #436: While loops not correctly highlighted

wimille, 2023-07-31:
**Describe the bug**
"While" loops are not highlighted. In the following example, "end loop" is correctly highlighted but not the "while (condition) loop".

**To Reproduce**
Write a simple HDL file with a while loop such as below.

**Code**
    process (clk, reset)
    begin
        if reset = '1' then
            c <= '0';
        elsif rising_edge(clk) then
            while ( a = '0' and b = '1' ) loop
                c <= '1'
            end loop;
    end process;

**Please complete the following information:**
 - OS: Win10
 - VSCode 1.80.2

wimille, 2023-08-09:
I don't know how an extension is made, or how the syntax files are used. 
However, i've take a look at vhdl.tmLanguage.json and in "loop_statement" i'dont understand why this part is not build as if or for statement. 
So i've made a simple modification to test:
i've replaced 

```    
 "beginCaptures": {
`           "1": {
              "name": "entity.name.tag.vhdl"
            },
            "2": {
              "name": "keyword.operator.vhdl"
            },
            "3": {
              "name": "keyword.reserved.vhdl"
            }
          },
```

by
```    
 "beginCaptures": {
`           "3": {
              "name": "entity.name.tag.vhdl"
            },
            "2": {
              "name": "keyword.operator.vhdl"
            },
            "1": {
              "name": "keyword.reserved.vhdl"
            }
          },
```

And now while loop seems to be correcty highlighted.

qarlosalberto, 2023-08-23:
I can't reproduce your problem. Can you share a screeshot?

wimille, 2023-08-24:
Hi,
here's a snapshot (with teros 5.0.10) : (it is a foo module, just for the example):
[image]