TerosTechnology/vscode-terosHDL #438: XHDL/XSIM Linter files in my local directory

pintert3, 2023-08-02:
I'm getting `xsim.dir` and `xhdl.pb` files in my local directory when I create a VHDL file. The issue seems similar to issue #98 . It's pretty annoying, cause they are created in every directory I create the files in.

**To Reproduce**
Set Vivado to be the linter and create a new project within which you create some VHDL files

 - OS: Windows 11
 - VSCode version 1.80.2

[image]

qarlosalberto, 2023-08-06:
Fixed in the new version (5.0.8). Feel free to re-open the issue if the problem continues.

Mystzl, 2023-08-07:
Still get this problem.
OS: windows 10
VsCode ver: 1.81.0
terosHDL ver: 5.0.9

qarlosalberto, 2023-08-07:
Can you double check it? Close VSCode, remove the Vivado files and re-open VSCode.

pintert3, 2023-08-07:
Ya, I can also confirm it's still there.
[image]

pintert3, 2023-08-09:
Still does it. 
[image]

qarlosalberto, 2023-08-09:
What is your OS? Only xsim.dir or also with the other files?

pintert3, 2023-08-09:
Windows 11. And yes only `xsim.dir` this time

pintert3, 2023-08-16:
Sorry, didn't reply in time, I must have forgotten to respond. It still happens. Only `xsim.dir` in every folder I open a vhdl file in.

qarlosalberto, 2023-08-17:
Try this version: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

I think that it's a bug in the windows delete command for directories

pintert3, 2023-08-21:
The `xsim.dir` still appears.

qarlosalberto, 2023-08-21:
I don't know what happen... If you type in the cmd: `del -Recurse xsim.dir`. Can you remove that folder? Because I'm running that command...

pintert3, 2023-08-21:
Right, so I tried that command: `del -Recurse xsim.dir` and it works....only in powershell. When I run it in the usual Windows CMD prompt, it doesn't work, it throws an error: 

```

C:\Users\Asus\Desktop\tests\teros\hello>del -Recurse xsim.dir
Could Not Find C:\Users\Asus\Desktop\tests\teros\hello\-Recurse
C:\Users\Asus\Desktop\tests\teros\hello\xsim.dir\*, Are you sure (Y/N)? y

```
And in the end, does nothing.

That gives me the idea that the command could be running in the command prompt shell instead of Powershell.
To remove it totally in the command prompt I'd use:
```
rmdir /s /q xsim.dir
```
And to do the same but in powershell, I'd use your command:
```
del -Recurse xsim.dir
```

qarlosalberto, 2023-08-21:
> So it doesn't work. but I've taken a look at the commit, and seen the
> command being run, in particular this part
>
> ... del ${path_3} && del -Recurse ${path_4} && Remove-Item -Recurse -Path ${path_4} && rmdir /s /q ${path_4}
>
> *TL;DR* : Replace the above snippet with
>
> ... del ${path_3} && rmdir /s /q ${path_4}
>
> Here's an explanation if you might need it:
> So for starters, I made a small test directory structure to test the
> command results. The directory structure is like so:
>
>     Directory: C:\Users\Asus\Desktop\mine\teros\test_deletion
>
>
> Mode                 LastWriteTime         Length Name
> ----                 -------------         ------ ----
> d-----         8/21/2023   5:57 PM                xhdl.dir
> -a----         8/21/2023   5:56 PM              0 xvlog.pb
>
> So of course I substitute the ${path_3} with xvlog.pb, which is a file,
> and ${path_4} with xhdl.dir which is a directory of course.
> Running the command with powershell brought me this:
>
> [... 24 lines cut ...]
> (versus ||), the last command doesn't run at all, which explains why only
> xhdl.dir remains.
>
> In conclusion, the solution is to remove all commands attempting to remove
> ${path_4} and replace them with rmdir /s /q ${path_4} .

Can you try to replace && by &? I need to do it compatible with both, CMD
and powershell

qarlosalberto, 2023-08-21:
I have moved it to a native implementation, plase check again: 

https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

pintert3, 2023-08-21:
It works perfectly now. Thanks!

qarlosalberto, 2023-08-21:
Finally! 😅