TerosTechnology/vscode-terosHDL #440: The simulation in the RUNS option is successful but cannot automatically jump out of gtkwave to display the wave file

xpww, 2023-08-03:
**Describe the bug**
A clear and concise description of what the bug is.
I installed iverilog and gtkwave in ubuntu, and then the simulation was successful in the RUNS option. The waveform file is in the ~/.teroshdl/build folder, but after the simulation is over, gtkwave cannot automatically jump out to display and read the waveform file.
**To Reproduce**
Code to reproduce the error.

**Code**
Code to reproduce the error.

**Please complete the following information:**
 - OS: [e.g.Ubuntu 18.04]
 - VSCode version version [e.g. 22]
-OS ubuntu22.04
-VSCode verison 1.81.0-insider
**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Additional context**
I tried different combinations in Configuration:Tools: execution mode and waveform viewer,such as :Command line/Gui + Tool Gui/GTKwave, but it also doesn't work.

tafkos, 2023-08-03:
I confirm also same issue on windows10 + conda env.

qarlosalberto, 2023-08-20:
It's not supported for iverilog.

marcos-bah, 2025-04-22:
Same issue here. Any updates or workarounds?

2025-04-21 23:23:29.130 [info] 
*************************************************
Checking general configuration (https://terostechnology.github.io/terosHDLdoc/docs/category/installation-checklist) 
🎉 Correctly configured 🎉🎉
*************************************************
⊙ Checking make. This tool is used to run some external tools. Current configured installation path: ""
  🔎 Searching for the binary "make" in the system path
  ✅ make Found in the system path using the command: "make --version"
🎉 The make installation path is correctly configured.

⊙ Checking Python installation. Configured installation path: ""
  ✅ Python was found in the path "python". Tried to find it using the command: "python -c "import sys; check_version = sys.version_info >         (3,0); exit(0) if check_version == True else exit(-1)""
🎉 Python installation path is correctly configured. Using "/usr/bin/python"

⊙ Checking Python dependencies. Current configured installation path: ""
  🎉 vunit found.
  🎉 edalize found.
  🎉 cocotb (optional) found.
  🎉 vsg (optional) found.
*************************************************
Checking Linter configuration (https://terostechnology.github.io/terosHDLdoc/docs/guides/linter) 
🎉 Correctly configured 🎉🎉
*************************************************
⊙ Checking linter errors with modelsim for VHDL. Current configured installation path: "/home/marcosbarbosa/intelFPGA/20.1/modelsim_ase/linuxaloem"
  🔎 Searching for the binary Modelsim at "/home/marcosbarbosa/intelFPGA/20.1/modelsim_ase/linuxaloem/vlog"
  ✅ Modelsim found at "/home/marcosbarbosa/intelFPGA/20.1/modelsim_ase/linuxaloem/vlog" using the command: "/home/marcosbarbosa/intelFPGA/20.1/modelsim_ase/linuxaloem/vlog -version"
🎉 The linter installation path is correctly configured.

⊙ Checking linter errors with modelsim for Verilog/SV. Current configured installation path: "/home/marcosbarbosa/intelFPGA/20.1/modelsim_ase/linuxaloem"
  🔎 Searching for the binary Modelsim at "/home/marcosbarbosa/intelFPGA/20.1/modelsim_ase/linuxaloem/vlog"
  ✅ Modelsim found at "/home/marcosbarbosa/intelFPGA/20.1/modelsim_ase/linuxaloem/vlog" using the command: "/home/marcosbarbosa/intelFPGA/20.1/modelsim_ase/linuxaloem/vlog -version"
🎉 The linter installation path is correctly configured.

⊙ Linter style for VHDL is disabled. Skipping configuration check.

⊙ Linter style for Verilog/SV is disabled. Skipping configuration check.

*************************************************
Checking External Tool Configuration (https://terostechnology.github.io/terosHDLdoc/docs/external_tools/) 
🎉 Correctly configured 🎉🎉
*************************************************
⊙ Selected external tool: MODELSIM. Installation path: "/home/marcosbarbosa/intelFPGA/20.1/modelsim_ase/linuxaloem"
  🔎 Searching for the binary modelsim at "/home/marcosbarbosa/intelFPGA/20.1/modelsim_ase/linuxaloem/modelsim"
  ✅ modelsim found at "/home/marcosbarbosa/intelFPGA/20.1/modelsim_ase/linuxaloem/modelsim" using the command: "/home/marcosbarbosa/intelFPGA/20.1/modelsim_ase/linuxaloem/modelsim --version"
🎉 The MODELSIM installation path is correctly configured.

⊙ Execution mode: CMD. The tool will be executend in the command line.

⊙ Waveform viewer: GTKWAVE. GTKWave waveform viewer will be opened after the simulation. Make sure that the GTKWave is correctly configured.
  🔎 Searching for the binary GTKWave at "/usr/bin//gtkwave"
  ✅ GTKWave found at "/usr/bin//gtkwave" using the command: "/usr/bin//gtkwave --version"
🎉 The Wavefrom Viewer installation path is correctly configured.

*************************************************
Checking Formatter configuration (https://terostechnology.github.io/terosHDLdoc/docs/guides/formatter) 
🎉 Correctly configured 🎉🎉
*************************************************
⊙ Checking formatter standalone for VHDL.
  The formatter is built into TerosHDL. Skipping configuration check.
🎉 The linter installation path is correctly configured.

⊙ Checking formatter istyle for Verilog/SV.
  The formatter is built into TerosHDL. Skipping configuration check.
🎉 The linter installation path is correctly configured.

*************************************************
Checking Schematic configuration (https://terostechnology.github.io/terosHDLdoc/docs/guides/schematic_viewer/installation) 
🎉 Correctly configured 🎉🎉
*************************************************
⊙ Selected backend: yowasp with installation path: "". This backend doesn't support VHDL files.
  🔎 Searching for the binary "yowasp" in the system path
  ✅ yowasp Found in the system path using the command: "yowasp-yosys --version"
🎉 The Schematic Backend installation path is correctly configured.