TerosTechnology/vscode-terosHDL #446: [Question] Configure paths relative to project root folder (linter Verilog include search path in particular)

eruanno123, 2023-08-10:
Hi,

Is it possible to set include search paths relative to the project root directory in linter settings? I failed to figure this out and I cannot find it in the documentation.

[image]

No problems if one uses absolute paths:

[image]

I tried `source/rtl/include`, but apparently the relative paths are literally relative to the file being parsed.

qarlosalberto, 2023-08-10:
Currently the paths are relative to the current file.

eruanno123, 2023-08-10:
I see. In that case, I would leave it as a feature request: The possibility of using some form of variables or string interpolation that would make resolving paths relative to certain project locations easy. 

TerosHDL uses its own configuration panel, so I don't know if VS code variables like `${workspaceFolder}` are even remotely possible. Therefore, how it could be implemented, I am leaving as an open question.