TerosTechnology/vscode-terosHDL #448: ghdl cannot find resource library vunit_lib

pintert3, 2023-08-17:
So while I know terosHDL uses VUnit, I'm stuck wondering how to write vunit based testbenches, if the linter itself can't find vunit_lib. Is there something I'm missing?
```vhdl
library vunit_lib;
context vunit_lib.vunit_context;

entity simple_tb is
    generic
    (
        runner_cfg : string
    );
end entity simple_tb;

architecture beh of simple_tb is
    signal clk     : std_logic := '0';
    signal reset_n : std_logic := '0';
    signal q       : std_logic := '0';

begin
    clk <= not clk after 10 ns;
    simple_i : entity work.simple
        port map
        (
            clk     => clk,
            reset_n => reset_n,
            q       => q
        );

    main : process
    begin
        test_runner(runner, runner_cfg);
        while test_suite loop
            ...
        test_runner_cleanup(runner);
    end process;
    test_runner_watchdog(runner, 10 ms);

end architecture beh;
```

qarlosalberto, 2023-08-17:
GHDL linter doesn't support external libraries. It's bette to use
standalone liner.

pintert3, 2023-08-17:
Oh, yeah I see. Thank you. Is there a reason for this though?

qarlosalberto, 2023-08-17:
Because I'm using `ghdl -s FILE_PATH`. This only allows 1 file. It's not trivial to consider more than 1 file.

closed 2023-08-18 by qarlosalberto