TerosTechnology/vscode-terosHDL #449: VUnit Project not runnable

pintert3, 2023-08-17:
I'm failing to run a sample vunit project. I've added sources from the vunit `run.py` file, I've made sure the inbuilt linter doesn't show any errors, but I'm still not getting any files in the **RUN** view. This is only happening for as long as I have **VUnit** selected in the Tools:General section. When it's **GHDL** it shows the top level tb in the **RUN** section when I select it. It's left me quite frustrated.

pintert3, 2023-08-17:
Just to add: this is on Windows 11, and on v5.0.10.
Also, I haven't found any documentation on running vunit projects on Teros ( I'm aware that it uses it under the hood though). Any help is appreciated.

qarlosalberto, 2023-08-17:
You need to add the run.py file and set it as top level of the project. It would be apreciate if you want to help me with the VUnit documentation.

pintert3, 2023-08-17:
For me, the run.py is set as the top level but it doesn't seem to affect **RUNS**.
[image]

qarlosalberto, 2023-08-18:
Have you checked that Python is correctly configured? https://terostechnology.github.io/terosHDLdoc/docs/getting_started/installation#validating-installation

pintert3, 2023-08-18:
Yes, it is, as far as I know:
[image]

qarlosalberto, 2023-08-18:
can you check the VUnit example from TerosHDL project samples?

pintert3, 2023-08-18:
I've tried it, set the run.py as current top level, but got the same results:
[image]

shrtique, 2024-01-05:
Add run.py to the sources and select it as a top file, then you should get a list in Runs

james-ziegler, 2024-08-15:
Was there a resolution to this? I am seeing the same problem. I have a VUint project running at the command line, but not in TesosHDL.

There are errors from the linter not being able to find the vuint library, is that at least part of the problem?

Thanks!