TerosTechnology/vscode-terosHDL #458: "Generate all project documentation" missing?

elpollodiablo84, 2023-08-28:
Maybe I'm wrong, but it seems to me that after the big v5 update the "Generate all project documentation"(*) feature is missing. If I'm not wrong, I would suggest to reintroduce the feature since it was really helpful.

(*): Just to be clear, I'm referring to the possibility to create a folder with the documentation of all the modules of the project, with hyperlinks between files and a main page with the project description.

qarlosalberto, 2023-08-29:
Yes, it's missing. I plan to do a refactor and I will add it in the future.

polyee13, 2023-09-20:
I'm also interested in this. Is it an alternative to the cli version? https://github.com/TerosTechnology/teroshdl-documenter-demo

qarlosalberto, 2025-03-18:
For now there isn't an alternative.

closed 2025-03-18 by qarlosalberto