TerosTechnology/vscode-terosHDL #472: VUnit based project runs error on Windows due to `mkdir -p`

pintert3, 2023-10-02:
**Describe the bug**
When trying to run a project whose sources are got from a vunit file, an error is thrown when trying to create directories:

```
2023-10-02 23:22:13.986 [info] "Creating libraries directories"

2023-10-02 23:22:13.989 [info] mkdir -p lib

2023-10-02 23:22:14.015 [info] mkdir -p vunit_lib

2023-10-02 23:22:14.038 [info] A subdirectory or file -p already exists.
Error occurred while processing: -p.

2023-10-02 23:22:14.042 [info] make: *** [make_libraries_directories] Error 1
```
I reckon the error is due to the whatever `make` command is being used, as it doesn't work directly on Windows.

**To Reproduce**
Create a VUnit `run.py`, a VHDL file and it's test bench.
Create and an empty project in TerosHDL.
Load source files to the project using VUnit run.py.
Set the testbench as the toplevel and try to run.

**Please complete the following information:**
 - Windows 11
 - VSCode version 1.82.3

pintert3, 2023-10-04:
For windows CMD and Powershell, the `-p` is not necessary

SebDunst, 2023-10-27:
Having the same issue here.
I'm also using Windows (Win10). I have everything perfectly installed (Check dependencies passes without any errors).

In my case I am trying a "GHDL"-Run with the "half_adder" example.
When compiling the two source files "half_adder.vhd" and "half_adder_tb.vhd" into library "work", then everything is fine.
But if I try to introduce some kind of library (e.g. compiling "half_adder.vhd" into "mylib", then there is the same error as described above.

Regards,
Sebastian

pintert3, 2023-10-27:
I did some digging earlier and realized it's an issue with **edalize**. I opened an issue there: [olofk/edalize#403](https://github.com/olofk/edalize/issues/403). Unfortunately, doesn't seem like they'll fix it soon, so I fixed mine by changing the 
`"mkdir -p {}".format(lib)` in the **edalize/ghdl.py** with `"if not exist {s} mkdir {s}".format(s=lib)`.

It fixed that particular bug for me, although I found another one related to ghdl failing to find the main vhd tb file I was attempting to simulate

olofk, 2023-12-08:
Sorry for being a bit slow here. Looking at the code I wonder if we need the `-p` switch at all. That would be required to create parent directories, but unless I'm misreading things, the command would always make a single-depth subdir in a known existing directory, right?

atoroSMT, 2025-02-11:
> Having the same issue here. I'm also using Windows (Win10). I have everything perfectly installed (Check dependencies passes without any errors).
>
> In my case I am trying a "GHDL"-Run with the [...]

Hello @SebDunst , did you find a solution for that? Thanks!

HuskyNator, 2025-11-01:
Strangely, I have this exact issue on one of my two windows devices.
Happens under the example half_adder example @SebDunst mentions.