TerosTechnology/vscode-terosHDL #474: Execution of vlib.exe failed (vlog-66)

voltavidTony, 2023-10-09:
I'm trying to get TerosHDL working, and sofar everything is good except for the linter. I'm using ModelSim, which I got through installing Intel Quartus Lite (So it's ModelSim-Intel FPGA Edition).
I use modelsim to run a design (vsim), but when used as the linter, it doesn't work. Instead, whenever I open a Verilog file (*.v), it just produces the error message "Execution of vlib.exe failed. Please check the error log for more details. TerosHDL: modelsim(vlog-66) [Ln 1, Col 1]". I don't know what this error log is, let alone where to find it. I already checked VSCode's log path, but there's nothing.
I've also read all the posts about needing a 'work' folder in the project root, which I have. All of the tools run correctly when I invoke them through the command line ('vlib work' runs correctly).
I don't know if it has anything to do with it, but the build process tries to invoke the 'vco' shell script which obviously doesn't work since my platform is Windows.

My system:
 - OS: Windows 11 22H2
 - VSCode version version: 1.83.0

DomininkDandaUofT, 2023-10-18:
I have this same problem, I'm honestly just trying to get started with this extension after having a terrible experience writing code in modelsim directly. I would appreciate if someone could elaborate on this bug!

qarlosalberto, 2023-10-18:
@DomininkDandaUofT also are you using Windows?

qarlosalberto, 2023-10-18:
In VSCode go to: Help -> Toggle Developer Tools -> Console. You will see the call to ModelSim. Please, copy and paste the output.

kiko1412, 2024-02-08:
> I have this same problem, I'm honestly just trying to get started with this extension after having a terrible experience writing code in modelsim directly. I would appreciate if someone could elaborate on this bug!

maybe you could see this
https://blog.csdn.net/zidan1412/article/details/135761059?spm=1001.2014.3001.5502