TerosTechnology/vscode-terosHDL #475: Make installation folder path:  Error: '['make']' exited with an error: 2

ziky0827, 2023-10-23:
**Describe the bug**
使用iverilog编译时出现错误
---> Build directory: C:Users79458.teroshdlbuild
---> Make installation folder path: 
Error: '['make']' exited with an error: 2

**Please complete the following information:**
 - VSCode version version 1.83.1

**Additional context**
Checking dependencies...
 Python found: python.exe
 Python found in path: D:\Program Files\Python 3.10.9\python.exe
 vunit found 
 cocotb found 
 edalize found 
 Make found in path: make.

Did I make a mistake in my configuration? Thank you for your reply.

11yuan, 2023-12-14:
我也遇到了同样的问题，请问你解决了吗？

qarlosalberto, 2023-12-14:
Set the developer mode and run it again. It should show more information:

[image]

ziky0827, 2023-12-14:
> 我也遇到了同样的问题，请问你解决了吗？

没有，后面就没用这个插件编译了

konosubakonoakua, 2024-07-11:
```
2024-07-11 23:01:29.147 [info] iverilog -sMAX2_tb -c test.scr -o test 

2024-07-11 23:01:29.186 [info] make: *** [Makefile:9: test] Error -1073741515

2024-07-11 23:01:29.189 [info] 
************************************************************************************************
---> Build directory: C:\Users\<myname>\.teroshdl\build
---> Make installation folder path: System path
Error: '['make']' exited with an error: 2
```

konosubakonoakua, 2024-07-11:
It's strange, If I run `make` in that folder manually, I got no error.
```
iverilog -sMAX2_tb -c test.scr -o test
c:\Users\<name>\Projects\verilog\learn\max2_tb.v:10: warning: Port 2 (num2) of MAX2 expects 8 bits, got 1.
c:\Users\<name>\Projects\verilog\learn\max2_tb.v:10:        : Padding 7 high bits of the port.
```

qarlosalberto, 2024-07-11:
It seems other error different of make. In TerosHDL configuration enable developer mode:

[image]

And run again.

konosubakonoakua, 2024-07-12:
already at developer mode