TerosTechnology/vscode-terosHDL #571: Documenter: VirtualBus direction not recognized

albydnc, 2024-01-15:
**Bug description**
When I specify the direction of the virtual bus using the keyword `@dir`, the documenter does not recognise it and puts it as a description of the bus.

**To Reproduce**
```vhdl2008
entity descrambler is
  port (
    clk_i       : in    std_logic;                                   --! clock
    reset_i     : in    std_logic;                                   --! active high
    --! @virtualbus Avalon-ST sink @dir in
    snk_data_i  : in    std_logic_vector(g_DATA_WIDTH - 1 downto 0); --! data input
    snk_valid_i : in    std_logic;                                   --! input valid signal
    snk_ready_o : out   std_logic;                                   --! ready backpressure to in signal
    --! @end
    --! @virtualbus Avalon-ST source @dir out
    src_valid_o : out   std_logic;                                   --! valid output signal
    src_ready_i : in    std_logic;                                   --! ready backpressure signal
    src_data_o  : out   std_logic_vector(g_DATA_WIDTH - 1 downto 0)  --! data output
  --! @end
  );
end entity descrambler;
```

**Please complete the following information:**
 - OS: Fedora 39
 - VSCode version 1.85.1
 - TerosHDL 5.0.12

**Screenshots**
[image]

qarlosalberto, 2024-01-15:
The description should be at the end:

```
    --! @virtualbus Avalon-ST @dir in sink
```

albydnc, 2024-01-15:
Thank you, this solved the issue.

closed 2024-01-15 by albydnc