TerosTechnology/vscode-terosHDL #574: The function of Autocomplete and snippets seems not work well on win11

narutozxp, 2024-01-27:
terosHDL does not autocomplete port signals when I work in win11， However, terosHDL works well when I connect my Ubuntu server by SSH extension.
Two environments have the same version of terosHDL(v5.0.12)
[image]
[image]

ckuhlmann, 2024-03-29:
I have the same problem on Win 10, TerosHDL v5.0.12 / VSC 1.87.2. I thought this was a limitation of TerosHDL not showing signals, but now that it seems to be a bug, I hope this gets fixed.
It is a rather severe limitation, if there's anything I can help to narrow the issue down, please let me know.

gmartina, 2024-04-13:
Hello @ckuhlmann @narutozxp , I tested it on W11 with v6.0.0 (pre-release) and it works. Please provide a minimal reproducible example (not an image).

[image]

ckuhlmann, 2024-04-21:
I haven't been able to pinpoint this issue, but one thing that usually causes it, is unsaved file changes. If you add a port or signal and don't save the file, you don't get the autocomplete suggestion for port-maps and inside procedures (inside the very same file). But saving the file and reopening the auto-complete list (ctrl+space), the port/signal is usually there after some time (few seconds). I think this is not due to the language server (using VHDL-LS), as the hover tooltip is available for me without saving the file. I'm not sure whether this is intended behavior / a known limitation, but since the language server works on the unsaved edits, it surely is unexpected from a user point of view.
Not sure whether the the Verilog side of the issue has this in common or is a totally separate issue.

narutozxp, 2024-05-10:
@gmartina Besides, your demo seems to show the function of vscode itself, not TerosHDL.