TerosTechnology/vscode-terosHDL #579: Quartus 23.1 fails to execute

kbleeke, 2024-02-17:
**Describe the bug**
I have Quartus Lite 23.1 installed. When I try to click the Compile Task, The following log message appears

```
2024-02-17 11:36:41.720 [info] 'C:\intelFPGA_lite\23.1std\quartus\bin64\quartus_ipgenerate' is not recognized as an internal or external command, operable program or batch file.
```

The path to the quartus installation is correct, however the "quartus_ipgenerate" program is missing.

**To Reproduce**
- Try to compile a project with Quartus 23.1

**Please complete the following information:**
 - OS: Windows 11
 - VSCode version version 1.86.2
 - Extension Version v6.0.0 prerelease

qarlosalberto, 2024-02-17:
Are you using TerosHDL 6.X ?

kbleeke, 2024-02-18:
Yes, I was using the most recent pre-release (as of yesterday) from the vscode marketplace.

qarlosalberto, 2024-02-18:
You need Quartus Pro >= 23.3

https://terostechnology.github.io/terosHDLdoc/docs/next/tool_manager/tools/quartus/getting-started

kbleeke, 2024-02-18:
The Pro version doesn't seem to support the MAX 10 (what I'm trying to target). But i guess that is Intel's fault.

closed 2024-02-18 by kbleeke