TerosTechnology/vscode-terosHDL #580: Python3 binary path setting not taken into account

mike6942069420, 2024-02-18:
I first open the global settings menu and go to the General page, then I set my python3 binary path to **/usr/bin/python3** and click "Apply".
When I then click on the "Check dependencies" button, I always see **Python found in path: /tools/Xilinx/Vivado/2020.2/tps/lnx64/python-3.8.3/bin/python3** and not the python interpreter at the path I specified it to. 

I am running the latest vscode version (1.86.2) on ubuntu 23.10 using python3 version 3.11.6.

I tried the release (v5.0.12) and prerealease versions of the teroshdl extension to no avail.

qarlosalberto, 2024-02-18:
does it work fine with the marker version, not pre release

mike6942069420, 2024-02-18:
Thanks for your response. No, it does not work with the marker version of teroshdl either.

qarlosalberto, 2024-02-18:
ok, I will take a look