TerosTechnology/vscode-terosHDL #590: Schematic viewer out put:[error] Yosys failed.

internet96, 2024-03-11:
**Describe the bug**
When I click to the Schematic viewer button, the out put is :

2024-03-11 12:53:09.710 [error] Yosys failed.
2024-03-11 12:53:09.716 [info]  yowasp-yosys -p "read_verilog -sv Control_Unit.v; ; proc;  ; write_json teroshdl_yosys_output.json; stat"

2024-03-11 12:53:09.729 [info] 'yowasp-yosys' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

[image]

I install anaconda-python3.11,make for windowsm,and all python packages it requires, please help me solve the problem. 
**To Reproduce**
Code to reproduce the error.

**Code**
Code to reproduce the error.

**Please complete the following information:**
 - OS: [e.g.Ubuntu 18.04]
 - VSCode version version [e.g. 22]

**Screenshots**
If applicable, add screenshots to help explain your problem. **Share the code as text, not as a screenshots!**

**Additional context**
Add any other context about the problem here.

arretxe, 2024-03-13:
I get the same problem in windows but I was assuming I was doing someting wrong until I saw this, now I don't know what to think anymore... :-S

arretxe, 2024-03-13:
> I get the same problem in windows but I was assuming I was doing someting wrong until I saw this, now I don't know what to think anymore... :-S

2024-03-13 08:15:14.166 [info] C:\Users\60146\Downloads\oss-cad-suite\bin\yosys.exe -m ghdl -p "ghdl --std=08 -fsynopsys c:\Users\60146\Documents\CAF\Vega\Sandbox\kkp\S.24.53.9999\Vega_HPL\FCPUD\SAFE_STATE\src\safe_state.vhd -e; ; proc;  ; write_json teroshdl_yosys_output.json; stat"

2024-03-13 08:15:14.486 [error] Yosys failed.

arretxe, 2024-03-14:
got it working on windows, just had to select the backend HDL+Yosys (instead of GHDL(module)+Yosys as suggested in the documentation.
However I noticed that it only works for vhd files which don't rely on packages

qarlosalberto, 2024-03-14:
Could you share how did you install/download ghdl?

arretxe, 2024-03-14:
> Could you share how did you install/download ghdl?

I used MSYS2 MINGW64 and installed from the package group mingw-w64-x86_64-eda package the mingw_w64-x86_64-ghdl-llvm and mingw-w64-x86_64-yosys

internet96, 2024-03-15:
> Could you share how did you install/download ghdl?

I just use Verilog but not VHDL，and I do it like that:
[image], 
[image]
 should I instll GHDL?

BOXin0405, 2024-04-29:
I meet the same problem, and have a try of adding the path of _yowasp-yosys.exe_ to the system path. 
(for example, C:\Users\xxx\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\Scripts)

It finally worked.

thuvasooriya, 2024-06-29:
faced the same issue i think it has to do with yowasp-yosys python activation.
modern python doesn't support system pip installations, so creating a venv is the way. but for some reason yowasp doesn't respect the first configuration of python. THIS SHOULD BE FIXED i guess.

for now passing `source ~/.pens/teroshdl/bin/activate &&` before yosys works

Niujingzhuo, 2024-08-09:
I've got the same problem :
ERROR: Can't guess frontend for input file `d:/*****.v;' (missing -f option)!

2024-08-09 14:10:21.994 [error] Yosys failed.
my dependencies are all installed

junlzu, 2024-09-28:
> I've got the same problem : ERROR: Can't guess frontend for input file `d:/*****.v;' (missing -f option)!
> 
> 2024-08-09 14:10:21.994 [error] Yosys failed. my dependencies are all installed

I'v got the same problem, too.