TerosTechnology/vscode-terosHDL #610: Schematic ERROR: Can't load module `./ghdl'

lithegreat, 2024-04-25:
**Describe the bug**
[info] ERROR: Can't load module `./ghdl': "C:\msys64\mingw64\bin\..\share\yosys\plugins\ghdl.so": �Ҳ���ָ����ģ�顣

I have intalled yosys and ghdl via [mingw-w64-x86_64-eda](https://packages.msys2.org/group/mingw-w64-x86_64-eda) as recommended, and added them to the PATH.

**Please complete the following information:**
 - OS: Windows 11 (Windows_NT x64 10.0.22631)
 - VSCode version version  1.88.1 (user setup)

**Screenshots**
[image]

lithegreat, 2024-04-25:
I solved it. You need to choose **Yosys + GHDL** instead of **GHDL(module) + Yosys** at Configuration.

qarlosalberto, 2024-04-25:
great! I will delete the bad option

hghuyho, 2024-09-16:
Hi, I had same issues. It doesn't work. Schematic Viewer backend: GHDL + Yosys (VHDL + Verilog/SV)
[image]

JakubFranek, 2025-02-16:
Hello @hghuyho, have you managed to solve the issue somehow?
Thanks
Jakub