TerosTechnology/vscode-terosHDL #612: Error finding raptor

MarkoMk, 2024-05-08:
I am trying to run a project and even though I se everyghing up like in the tutorials on the webpage I am getting the following error:
/Raptor/build/bin/raptor: not found

Is it possible that TerosHDL only works with raptor version 2023.04 and I curently have version 2024.05?

MarkoMk, 2024-05-08:
[image]

qarlosalberto, 2024-05-08:
Can you type "/Raptor/build/share/raptor --version" in the terminal and to share a screenshot?

MarkoMk, 2024-05-08:
[image]

qarlosalberto, 2024-05-08:
Please double check your configuration in TerosHDL. I think that the binary path it's not correct. And check that your are using the complete path

MarkoMk, 2024-05-08:
I think writing the bin path like this: ~/Raptor/build/bin fixed the issue with being unable to find the raptor shell script. Not getting that error anymore.

qarlosalberto, 2024-05-08:
Great! I close the issue.

closed 2024-05-08 by qarlosalberto