TerosTechnology/vscode-terosHDL #613: vscode extension TerosHDL doesn't work after installation in vscode

mrBitman, 2024-05-15:
Hello

I need help

I downloaded your vscode extension (teros-technology.teroshdl)

But out the box just "Module documentation preview works"

At the vscode Output report i see:
`2024-05-15 15:59:27.155 [error] Traceback (most recent call last):
  File "/Users/21151224/.vscode/extensions/teros-technology.teroshdl-5.0.12/node_modules/teroshdl2/out/project_manager/dependency/vunit_dependency.py", line 28, in <module>
    import vunit.project as pj
ModuleNotFoundError: No module named 'vunit'`

When i pick "Schematic viewer" button, at the vscode Output i see:

`2024-05-15 16:03:24.557 [info]  yowasp-yosys -p "read_verilog -sv test.sv; ; proc;  ; write_json teroshdl_yosys_output.json; stat"

2024-05-15 16:03:24.712 [info] 

2024-05-15 16:03:24.712 [info]  /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |

2024-05-15 16:03:24.714 [info]  |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.36+8 (git sha1 fb4cbfa73, ccache clang 14.0.0-1ubuntu1.1 -Os -flto -flto)

-- Running command `read_verilog -sv test.sv; ; proc;  ; write_json teroshdl_yosys_output.json; stat' --

1. Executing Verilog-2005 frontend: test.sv

2024-05-15 16:03:24.718 [info] test.sv:15: ERROR: syntax error, unexpected TOK_ID, expecting '(' or ';' or '#'

2024-05-15 16:03:25.254 [error] Yosys failed.`

and this previewer window is blank.

Help me

Thank you

mrBitman, 2024-05-15:
[image]

mrBitman, 2024-05-15:
**System Info:**

Version: 1.85.2 (Universal)
Commit: 8b3775030ed1a69b13e4f4c628c612102e30a681
Date: 2024-01-18T06:40:32.531Z (3 mos ago)
Electron: 25.9.7
ElectronBuildId: 26354273
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Darwin arm64 23.4.0

qarlosalberto, 2024-08-31:
```
 2024-05-15 15:59:27.155 [error] Traceback (most recent call last): File "/Users/21151224/.vscode/extensions/teros-technology.teroshdl-5.0.12/node_modules/teroshdl2/out/project_manager/dependency/vunit_dependency.py", line 28, in <module> import vunit.project as pj ModuleNotFoundError: No module named 'vunit'
```

You need to install VUnit dependencies and to configure Python: https://terostechnology.github.io/terosHDLdoc/docs/getting_started/installation#requirements

```
2024-05-15 16:03:24.718 [info] test.sv:15: ERROR: syntax error, unexpected TOK_ID, expecting '(' or ';' or '#'
```

It seems you are using not synthesizable code.

closed 2024-08-31 by qarlosalberto